WaresWichall/hass-anycubic_cloud #1: Can't login

lucagomm, 2024-07-22:
The first time I tried it, the login worked and I could see my printer as a checkbox entry, but I could not submit it because only the checkbox was removed, but the printer was not added. Since then I can't even log in and get the message that my login details are incorrect.

WaresWichall, 2024-07-22:
Hi @lucagomm I've just pushed a lot of updates can you do a fresh download of the component and try again?

Anycubic servers have been down a lot lately but today they do seem stable...

If you add the following to your home assistant `configuration.yaml` then you should get extra logs from the component which will help me diagnose the issue:

```yaml
logger:
  logs:
    custom_components.anycubic_cloud: debug
```

WaresWichall, 2024-07-22:
If you're in the homeassistant or anycubic discord we can chat there too

WaresWichall, 2024-08-11:
Hi @lucagomm there's been lots of updates since you opened this issue and I've had reports from a few users that Kobra 2s are working now so I'm closing this, feel free to open a new issue if you have problems with the latest version

closed 2024-08-11 by WaresWichall