WaresWichall/hass-anycubic_cloud #8: Layers

matuck, 2024-08-08:
The current layer and total layers is not reporting correctly.  The current layer is reporting as 0 and total layers is -1.  This is on a kobra 2 pro.

Thank you
This is the interface I have been looking for.

matuck, 2024-08-08:
Doing some additonal looking this may not be able to be fixed.  Looking in the app. The layer isnt reported either.

WaresWichall, 2024-08-10:
It's possible the layer isn't reported for the kobra 2s, are you on the anycubic discord?

If you download the diagnostics for the anycubic cloud integration you can send it to me so I can have a look at the data. PM on discord is probably best otherwise I suggest having a look through the file before you upload it here and delete anything personal. For example it should give you a list of projects in JSON but I only need the latest one to have a look.

[image]

WaresWichall, 2024-08-10:
I'm also on reddit as wcubic if you want to PM there

WaresWichall, 2024-08-25:
We're still looking into this but just to let others know - it does look like this info isn't available for Kobra 2s at the moment