WaresWichall/hass-anycubic_cloud #10: Resin printer support

AurelioB, 2024-08-14:
I've got an M7 Pro. I managed to setup the integration with my anycubic account but unfortunately it crashes when dealing with my printer (and I guess with any resin printer) probably because it's got no nozzle. This is the error I'm getting:

> 'NoneType' object has no attribute 'curr_nozzle_temp'

I understand that resin printers are very different to FDM printers, but maybe they could also be simpler to support (less stuff to check)

I'd be glad to help debug this if needed

WaresWichall, 2024-08-14:
Aha I was waiting for the first resin user to come along :D

Yeah there will be loads of errors for sure, but you say you managed to add the config to home assistant?

If so you should be able to download the diagnostics for the integration and then if you can send me that I can work through stuff. Will also need to work out the resin MQTT data too but step 1 first :)

Are you on the anycubic discord? That's the easiest way to send me logs in PMs I think

adamoutler, 2024-08-24:
It works for me on M5s.  But lots of unpopulated sensors. 

[image]

adamoutler, 2024-08-24:
File support is non-exisxtant

adamoutler, 2024-08-24:
[image]

adamoutler, 2024-08-24:
Hello, I made the original Anycubic Wifi domain for the m5s.  I'm very glad to see support for the Anycubic Cloud and not just wifi anymore!   This is great and allows me to get notifications when my printer is done.   Thank you!

There are some problems I've encountered that I want to report. 

Current prerelease version displays sensors which do not exist on the Mono M5S resin printer
[image]
[image]

File support does not work with m5s resin
[image]

Logs here: https://hackedyour.info/EhWduy775flNkCee

See here for more constructive feedback. https://community.home-assistant.io/t/anycubic-cloud-component-frontend-card/742295/27?u=adamoutler

adamoutler, 2024-08-24:
Mqtt-related message https://hackedyour.info/D8bPGY8dTrX3NOa4

WaresWichall, 2024-08-25:
I've just released 0.0.13 with initial resin printer support :)

I need to do more work to add resin print settings to the `Print Settings` modal still.

I also will need from one of you a home assistant log for the duration of a print with the `Debug Mode` enabled in the anycubic cloud integration, you'll also need to change the MQTT connection mode to at least printer online - or always on.
And lastly you'll need the following in your home assistant `configuration.yaml` to get the full logging of the component:

```yaml
logger:
  logs:
    custom_components.anycubic_cloud: debug
```

Restart home assistant afterwards and then start a print with a fairly fresh log - it will be best to DM me your log not post it here and feel free to remove anything not anycubic related from it