WaresWichall/hass-anycubic_cloud #13: Print (no Cloud Save) or Print( Save in User cloud

Sacx83, 2024-08-20:
Thank you for the Intergration been looking for one since last year,
[image]
  on the print no cloud or print  save in user cloud  there is not drop file   section  in any of the printers  I have a Kobra 2 pro and a Kobra 3

WaresWichall, 2024-08-21:
What version of home assistant is this?

Sacx83, 2024-08-21:
Core
2024.8.2
Supervisor
2024.08.0
Operating System
13.0
Frontend
20240809.0

WaresWichall, 2024-08-21:
Do you see any errors in the browser console?

I'm not on 2024.8 yet but I think I've seen other people without this issue on 2024.8

WaresWichall, 2024-08-21:
I've just pushed some changes to the `dev` branch which may fix this, can you try it out?

Let me know if it's not easy for you to download and I can make a pre-release

Sacx83, 2024-08-21:
ok I will

Sacx83, 2024-08-21:
can you do a pre realease pleaseeasier for me

WaresWichall, 2024-08-21:
Try this https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.0.13rc1

Sacx83, 2024-08-21:
Thank you I will let you know

Sacx83, 2024-08-21:
Yes still the same 
[image]
I reinstall everything with the new version installed the printer one by one and still the same  Ill be keep trying to figure out on my side

WaresWichall, 2024-08-21:
When I have time to update to 2024.8 I'll be able to look myself.

Can you try expanding more detail on the error in the browser console and sending a screenshot of it?

WaresWichall, 2024-08-21:
I think I've fixed this @Sacx83 
Try this https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.0.13rc2

Sacx83, 2024-08-21:
hooray!!!! its working  after I download the 13.rc2  and restart didnt show up but I deleted the cache and the history of Chrome and worked pefect thank you..
Now lets move to the multiprinting lets give it a test

WaresWichall, 2024-08-21:
Great! :) thanks

Sacx83, 2024-08-21:
Thank you very much for your help~

closed 2024-08-21 by Sacx83