WaresWichall/hass-anycubic_cloud #14: Multiple Printers

Sacx83, 2024-08-21:
Hey there So  i test the Integration with multiple Anycubic Printers Kobra 3 and Kobra 2 Pro, there was a problem
**Kobra 2 Pro**
The percetage is not working 
Project state is always on Fail
**Kobra 3**
Everything works fine

Working with both printers at the same time
Kobra 3 overwrite everysingle status on Kobra 2, Time Project name ,Percentage everything get overwrited will add some pictures later.
Thank you

WaresWichall, 2024-08-21:
Are you able to try out the latest changes on the `dev` branch and see if that resolves it for you?

Sacx83, 2024-08-21:
Let me take a look

WaresWichall, 2024-08-21:
Try the latest pre-release for this too

Sacx83, 2024-08-21:
Ok doing that right know will let you know

Sacx83, 2024-08-21:
[image]
Well Well Well its working  as you can see on the screenshoot one thing i did  find i could do was switching speeds on the kobra 2 in the kobra 3 works fine but not on the kobra 2

WaresWichall, 2024-08-21:
Brilliant :)

Can you usually change the kobra 2 speed in the app?

Can you show me a pic of the speed dropdown options when you click `Print Settings`?

Sacx83, 2024-08-21:
yes I can
[image]
[image]

WaresWichall, 2024-08-21:
Sorry I meant when you click on the speed dropdown there where it says Standard

Also does it show `Standard` for both printers?

Sacx83, 2024-08-21:
[image]
Actually it did not change on any of the printers  I will keep trying tomorrow I have to get back to work lol

WaresWichall, 2024-08-21:
No worries let me know :)

WaresWichall, 2024-08-21:
I pushed a pre-release to the card and this repository which may fix the speed bug, let me know

Sacx83, 2024-08-21:
I will test tomorrow  morning  and  I will keep you post it thx a lot

Sacx83, 2024-08-22:
Well, I just tested both of the printers a they works great!!! fantastic job thank you

PR #15 (2024-08-25): v0.0.13 - Bug fixes for multiple printers, Home Assistant 2023.8, Resin printers
# v0.0.13 Changes

- Bug fixes for multiple printers in frontend and integration
- Bug fixes for Home Assistant 2023.8 service -> action changes
- Bug fixes for multiple printers, closes #14
- Initial support for resin printers, more work needed, tracking in #10
closed 2024-08-25 by WaresWichall