WaresWichall/hass-anycubic_cloud #20: Anycubic API

pzy, 2024-09-04:
Not a real issue but i just really wanted to thank you for the repository, i borrowed the API Code for some private Project where i attached an LED Strip to my printer utilizing the API to show some status information. Which leads to my main point, i just figured out how to use the API Classes myself and so on, but i guess it would be helpful if this would be a seperate repository and probably a bit of description how it can be used.

WaresWichall, 2024-09-08:
I did have a future plan to strip the api out to a python library, but all my big future plans are kind of on hold right now as I'm a bit worried about Anycubics continued attempts to break this and their total lack of response to me.

I've already put in a lot of work and I don't want to keep putting more in just to have them shut it all down ...

I'm hoping I might get a response next week, we'll see :)

Cloud500, 2024-09-23:
I'll keep my fingers crossed for you (And honestly for “us”), the API is great (use it for external control and sync of various settings outside the enclosure)

It would be a shame if Anycubic locks externally access, especially with regard to the “Craft thoughts” of 3D printing...

WaresWichall, 2024-09-26:
I still haven't had a response from Anycubic, I don't expect to anymore... (:

I might create a python package just for the API later this year depending on how things unfold, but it is a bit much for a one man job :) ...

I will update the demo scripts soon to use the new token logon method but sounds like you've both worked that out anyway ;)

I did notice the bambulab integration for HA seems to also be using an internal api lib and they have a whole bunch of contributors over there :D

Cloud500, 2024-09-27:
I have almost no idea about the Homeassistant, I'm just testing/play with it for the first time, but if you extract the API as a package, I can certainly help/support you, Python is almost my "mother tongue" ;-)

WaresWichall, 2024-10-09:
@Cloud500 @pzy 
I've just finished a fairly big refactor of the codebase to annotate everything and improve error handling :)

Still testing out for bugs but I've pushed it to the [dev](https://github.com/WaresWichall/hass-anycubic_cloud/tree/dev) branch