WaresWichall/hass-anycubic_cloud #23: Invalid user token

Julinator05, 2024-09-19:
The integration worked perfectly for me until today (September 19, 2024), but today I got the error message that the integration had to be reconfigured. I have not logged in to my slicer or anywhere else. I then tried the installation steps again several times, but always got the error message “Invalid user token”. 
I have installed the following versions:
anycubic kobra 3: 2.3.3.9
anycubic color engine pro: 1.3.82
Homeassistant: 2024.7.2 (docker) --> then I did an update: 2024.9.2 (docker), problem is still there

I find the following error output in the log:
2024-09-19 19:33:56.152 ERROR (MainThread) [custom_components.anycubic_cloud] Invalid token, expected 238 chars, got 236.

and no, I did not copy the '', I have already double-checked that.

Does anyone else have the problem or knows a solution?

WaresWichall, 2024-09-19:
@Julinator05 please try the beta pre-release

Julinator05, 2024-09-19:
@WaresWichall Thanks a lot! It worked. I really appreciate your work!

WaresWichall, 2024-09-19:
No worries :) made a proper release with it now but it's the same as the pre-release you have

closed 2024-09-19 by WaresWichall

KaiOppi, 2024-09-20:
@WaresWichall with the 0.1.2 i get the same error. Invalid User Token. I try it today

WaresWichall, 2024-09-20:
@KaiOppi this is a different error can you please open a new ticket and paste the full error message

Julinator05, 2024-09-20:
> @WaresWichall with the 0.1.2 i get the same error. Invalid User Token. I try it today

@KaiOppi I got the v0.1.2rc1, maybe that will help

PR #24 (2024-09-19): v0.1.2 - Minor bug fixes
# v0.1.2 Changes

- Fixes #23 - user token validation 

Reminder: v0.1.0 included breaking changes which will affect you if upgrading from an earlier version
See #19