WaresWichall/hass-anycubic_cloud #25: Invalid user token.

KaiOppi, 2024-09-24:
When i try to add the addon to my HA i got this error. I try different versions from the Repo.

Logger: custom_components.anycubic_cloud
Quelle: custom_components/anycubic_cloud/config_flow.py:135
Integration: Anycubic Cloud
Erstmals aufgetreten: 08:56:43 (3 Vorkommnisse)
Zuletzt protokolliert: 08:57:29

Invalid token, expected 236 or 238 chars, got 238.

WaresWichall, 2024-09-24:
@KaiOppi can you try the latest beta pre-release?
https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.1.3rc1

KaiOppi, 2024-09-24:
@WaresWichall thanks, works! the message that i have to restart HA dont show up. After i install and restart the 0.1.3rc1 it works

WaresWichall, 2024-09-24:
Great thanks for letting me know @KaiOppi ! Will push a new release later.

Do you see any warnings in your logs for anycubic?

PR #26 (2024-09-25): v0.1.3 - Minor bug fixes
# v0.1.3 Changes

- Fixes #25 - user token validation 

Reminder: v0.1.0 included breaking changes which will affect you if upgrading from an earlier version
See #19
closed 2024-09-25 by WaresWichall