WaresWichall/hass-anycubic_cloud #30: Translation

MaxLucien, 2024-09-27:
Today i've tried to make some translations to German. After i added de.json in Anycubic Cloud ans Frontend Card, there waa no Option to change to Language, so i've tested it by translate the en.json directly. Then i've seen there are also Phrases within some other Files where are no Vars set for Translation.

Will they be added ? 

greetings Mike

WaresWichall, 2024-09-28:
@MaxLucien can you open a `draft` PR with your new `de.json` and I will work through this with you.

Or you can just post the new json here if you like

Can you let me know which phrases are missing too?

WaresWichall, 2024-09-28:
@MaxLucien if you do open a draft PR base it on the `dev` branch

WaresWichall, 2024-11-03:
@MaxLucien I've finished pushing some updates refactoring localisation using `%key` variables similar to home assistant, making it a lot less work to translate the strings.

This is all still on the `dev` branch for now

The integration language file is now a lot more compact:

https://github.com/WaresWichall/hass-anycubic_cloud/blob/dev/custom_components/anycubic_cloud/translations/input_translation_files/en.json

And the frontend language file should include all frontend strings:

https://github.com/WaresWichall/hass-anycubic_cloud/blob/dev/custom_components/anycubic_cloud/frontend_panel/localize/languages/en.json

Have a look at my notes here if you want to try building:
https://github.com/WaresWichall/hass-anycubic_cloud/blob/dev/DEVELOPMENT.md