WaresWichall/hass-anycubic_cloud #31: [enhancement] Filament consumption of Project

Cloud500, 2024-09-28:
I have seen that the API outputs the filament used for printing in the projects under `slice_material_info_list`, it also matches the slicer's filament usage (anycubic slicer next) so seems to be accurate.
_(Not checked yet, but it's a list, I assume one entry per filament used for Multi color.)_

Would it be possible to include this information in the HA integration?

Background:
I use Spoolman to manage my spools and with location and an extra field (ACE Pos.) I already have a way to specify the consumption  "manually" via a local Python script when slicing/starting to simplify the whole thing.

The idea/plan would be to automatically update the spools in Spoolman when a print is started.
(At least I've been able to create an automation that triggers when the printer starts ;-) )

WaresWichall, 2024-09-28:
@Cloud500 how would you like this info to be available?

That info is not always available, if I remember rightly it's not available for locally started jobs - only for stuff uploaded/printed via the cloud... You can see this behaviour in the mobile app too with less info available for local jobs

What is probably the most correct HA way for me to do this would be to fire an event tagged as `anycubic_cloud` when you start a print via HA with all that data for you to then use in automations with an event trigger and do what you want with the data

Cloud500, 2024-10-02:
Thanks, I'll have a look tomorrow (I just read `material_list` in #27 and was looking forward to it ^^)

The delay is also an idea, I thought about it at the beginning, but then decided to use the “switch off when finished” switch. This is the most practical solution for me, as it is the exception (as long as the enclosure is not completely finished) that he runs “unobserved”.

Cloud500, 2024-10-03:
Looks pretty good so far, except for a little confusion on my part
```
Filament usage from Slicer 24,15g
Print supplies usage 7842 in HA
```
It took me a bit (too long :satisfied: ) to realize that the value is in mm.
`Filament usage from Slicer: 7,72m`
(Does not quite fit, with 122mm difference, but if I add the poops and consider the probably more accurate measurement of the extruder I would assume that it fits so far)

What really excites me is that it is a “live” usage, so better than expected, especially for “broken” prints.

I've only tested it with Singe Filament so far.
I'll test with multiple Filaments this afternoon/tomorrow and see what it outputs then.
(I hope that I get a value for each filamet used)

The one thing that is somehow missing is an assignment of which filament it is, i.e. the ACE Pos. ID.
Especially with automation.

Either way, great work, thank you very much for it :+1:
I'm excited to see what else is possible

WaresWichall, 2024-10-03:
Haha some things are not up to me unfortunately :) the new `print_supplies_usage` attribute is straight from the anycubic 'project' API - I think it is just a total value not separate, and yes apparently in `mm` :)

The new `anycubic_cloud` event which is fired when you start a print DOES show individual totals, but it's not updated throughout a print

Cloud500, 2024-10-03:
I thought that might be the case, so the choice is between live consumption of (all) material or total consumption of individual filaments for the print...
(Ok, we could still calculate from the gcode itself, but ... do we really want that? :sweat_smile:)

For personal interest only the last one is of use to me as my goal is to automatically transfer consumption to Spoolman.
(This is no longer accurate if the print fails, but it's better than always doing it manually)
Both in combination are of course great for single filament prints (first for the consumption and time for the identification)

PS:
I have monitored the last prints and the results are mixed:
```
Slicer: 1540mm 
HA: 1661mm 
Diff: 121mm

Slicer: 1500mm 
HA: 1617mm 
Diff: 111mm

Slicer: 36700mm
HA: 23822mm
Diff: 12878mm
```
(Unfortunately I don't have a digital scale with corresponding scaling, otherwise I could actually check it)

WaresWichall, 2024-10-04:
@Cloud500 I've added all 3 available units to the start print event and also added a new example script to parse gcode files I thought might be useful for you :)

https://github.com/WaresWichall/hass-anycubic_cloud/tree/dev/custom_components/anycubic_cloud/scripts#gcode-metadata-dump

https://github.com/WaresWichall/hass-anycubic_cloud/blob/f54aaca0fe4a1d5b9a77fc331bdf67ada7b8c454/custom_components/anycubic_cloud/scripts/gcode_metadata_parser.py#L40

PR #27 (2024-10-21): v0.2.0 - Improvements to integration + card
# v0.2.0 Changes

## General
- Updates for Home Assistant 2024.8 & 2024.9
- Add option to config flow for yaml card config to customise panel printer card
- General codebase refactor to annotate and add more error handling for server issues
- Better handling of job sensors / api functions for:
  - `Job State` sensor + `print_status` project property (new states added)
  - `Job In Progress` sensor + `print_in_progress` project property
  - `Job Paused` sensor + `print_is_paused` project property
- Improved job preview image filling for locally started [... cut]
PR #35 (2024-10-23): v0.2.0 - Improvements to integration + card
# v0.2.0 Changes

## SEMI BROKEN BY ANYCUBIC
- See here: https://github.com/WaresWichall/hass-anycubic_cloud/issues/33

## General
- Updates for Home Assistant 2024.8 & 2024.9
- Add option to config flow for yaml card config to customise panel printer card
- General codebase refactor to annotate and add more error handling for server issues
- Better handling of job sensors / api functions for:
  - `Job State` sensor + `print_status` project property (new states added)
  - `Job In Progress` sensor + `print_in_progress` project property
  - `Job Paused [... cut]
closed 2024-10-21 by WaresWichall