WaresWichall/hass-anycubic_cloud #32: Uploading gcode: Validation error: Could not read gcode file.

duckida, 2024-09-28:
I have to click the Print button twice (once it says Validation error: Could not read gcode file.) and it works the second time. I have Kobra 2 Pro

TheHolyRoger, 2024-09-28:
@duckida how long are you waiting between presses?

The error message is likely coming from the second press, it's just that the first press takes a while to go through.

I should disable the button while it's processing though

duckida, 2024-09-28:
I press it once. It fails. Then I wait 2 seconds and press again, Then it works. Sometimes it takes more that 2 tries.

WaresWichall, 2024-09-28:
Hey @duckida can you try [v0.1.4rc3](https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.1.4rc3) and let me know if you get any extra error warning when the file upload fails

duckida, 2024-09-28:
How do I install that with HACS?

WaresWichall, 2024-09-28:
@duckida if you enable beta/pre-releases you should see that as an option to download then restart

duckida, 2024-09-28:
Its only showing rc2

duckida, 2024-09-28:
Is it under redownload?

WaresWichall, 2024-09-28:
1 sec I just saw a typo, I'll push a new release

WaresWichall, 2024-09-28:
@duckida you should see a button to "update information" - do that and then download `v0.1.4rc4`

PR #27 (2024-10-21): v0.2.0 - Improvements to integration + card
# v0.2.0 Changes

## General
- Updates for Home Assistant 2024.8 & 2024.9
- Add option to config flow for yaml card config to customise panel printer card
- General codebase refactor to annotate and add more error handling for server issues
- Better handling of job sensors / api functions for:
  - `Job State` sensor + `print_status` project property (new states added)
  - `Job In Progress` sensor + `print_in_progress` project property
  - `Job Paused` sensor + `print_is_paused` project property
- Improved job preview image filling for locally started [... cut]
PR #35 (2024-10-23): v0.2.0 - Improvements to integration + card
# v0.2.0 Changes

## SEMI BROKEN BY ANYCUBIC
- See here: https://github.com/WaresWichall/hass-anycubic_cloud/issues/33

## General
- Updates for Home Assistant 2024.8 & 2024.9
- Add option to config flow for yaml card config to customise panel printer card
- General codebase refactor to annotate and add more error handling for server issues
- Better handling of job sensors / api functions for:
  - `Job State` sensor + `print_status` project property (new states added)
  - `Job In Progress` sensor + `print_in_progress` project property
  - `Job Paused [... cut]
closed 2024-10-21 by WaresWichall