WaresWichall/hass-anycubic_cloud #33: Anycubic Cloud MQTT access blocked

WaresWichall, 2024-10-20:
I have bad news for everyone using this.

At some point recently Anycubic updated their servers to block MQTT access for our "ac_web" tokens and only allows MQTT access via app/slicer tokens now.

This change was specifically targeting this home assistant integration since the basic anycubic web client doesn't need or support MQTT connections.

Having put many hours of work into this integration and witnessed Anycubic continually try to break it with questionable updates, now that I know for sure they're targeting this integration I've had enough and I'm moving on.

I'm tired of the way they treat their customers and all of my tickets/emails about API access have been ignored.

I will publish a final release soon with an option to disable MQTT connections, meaning the integration will only get updates every minute - until Anycubic shut that down too.

Again, sorry for anyone using this but I'm selling my printer and getting something else that doesn't try to harvest all my data.

adamoutler, 2024-10-21:
I feel like there is a lot left to explore.  Let's talk about the various pathways to getting info. 
* MQTT - Off limits. Personally I wouldn't want to manage multiple users in an MQTT server. There are too many problems which could occur.
* SSH - we have the roof username/password for some of the units but not all.  Once in, we can access the `comm` binary and system logs as well as the internal system UARTs.  There is certainly a key to MQTT there, but again, Anycubic probably doesn't want that. It adds a level of risk to the device as well where a local attacker could perform a denial of service on your printer or fire the lamp until dead. I'd expect this to be patched
* Telnet - same as above. 
* WebUI - some units use OpenWRT and we have the user/pass but it grants access to ways to obtain access.
* Port 11311 - what is this?  I don't know, but `comm` opens it. When you connect to this port it disconnected the device from AC Cloud and each message sent can be observed as received in the `comm` logs.  After disconnect, the unit reconnects to AC Cloud within about 20s.

So based on this we can assume port 11311 is an alternative command and control function which allows local access to the device.  Additionally it looks to be an undocumented feature, less problematic for AC than a MQTT connection, and supports SOME feature set using Telnet.  

I don't have the time to reverse engineer this feature.  I don't have any examples of it being used locally for setup/control/monitoring.  If anyone knows, please reach out so that I can make something open source that can be used to tell us when our printers are finished.

Cloud500, 2024-10-21:
Just back from vacation and now a message like this :cry:

I had just set up my dashboard in a more or less satisfactory way and thanks to AppDeamon I was also able to automatically transfer the filament consumption to Spoolman :sob:

[image]

Maybe there is still hope for the LAN mode...
Unfortunately, I don't have time for this at the moment either, but I'll see if I can run the LAN mode from time to time and see what happens "n the network".

WaresWichall, 2024-10-21:
My understanding is that each type of printer handles local connections entirely differently with no unified API, so each model would need its own implementation.

The only thing that is the same on all their printers is that they connect to the cloud. So my suggestion to anyone that wants to focus on this would be to create a hassio docker addon emulating the cloud similar to this project: https://github.com/anjomro/kobra-unleashed 

That could then be the "broker" to talk to a modified version of this home assistant integration, as long as there is a way to force each printer model to use a local broker.

I won't be taking part in that though I'm afraid.

WaresWichall, 2024-10-21:
I made a pre-release with an option to "never connect" to MQTT to at least keep warnings out of the logs

https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.2.0rc8

I won't bother making a full release or finalising the notes since all the MQTT functionality is now broken...

adamoutler, 2024-10-21:
It seems like the future devices will be using `comm` binary.  I'm going to need to check out that project you suggested.  I'd really like some help with this port 11311 thing too. I can see OpenWRT being a passing fad, but it seems like comm is here to stay and comm has the port 11311 connection feature.  Maybe if we can figure out what it needs, it may be a viable option for local communication.

Cloud500, 2024-10-22:
Maybe there is another possibility, a completely different approach, but 2 things are interesting:

1. Anycubic seems to have released the firmware OpenSource (For the Kobra 3)
(https://github.com/ANYCUBIC-3D/Kobra3)

3. there is a first (probably quite good) working CFW
(https://github.com/utkabobr/DuckPro-Kobra3)

(And if nothing works, maybe we can connect an ESP directly to the UART :satisfied:)

adamoutler, 2024-10-22:
That's a microcontroller version but I do notice the command STATS seems to be the one I'd be interested in.

Cloud500, 2024-10-22:
The more approaches, the better the chances ;-)

duckida, 2024-10-23:
Is there a sniffer or something to get tokens out off the app or slicer? They are just web apps it seems.

duckida, 2024-10-23:
The app link for the sign in seems to be http://anycubic-1307326732.cos.ap-guangzhou.myqcloud.com/, mqtt-universe.anycubic.com, or uc.makeronline.com

WaresWichall, 2024-10-23:
@duckida the slicer usually stores tokens in a file in the config directory but it uses a slightly different login method which would need to be supported in this code.

The next problem is that Anycubic only permits one login per token which means that logging into the slicer would log you out in home assistant and vice versa

duckida, 2024-10-23:
Then don’t use the slicer

WaresWichall, 2024-10-23:
@duckida I'm not using anycubic anymore... :) just wanted to share the info I know

duckida, 2024-10-23:
Not sure but it might be https://cloud-platform.anycubicloud.com/Login?redirect=%2Ffile

duckida, 2024-10-23:
Do you know where in the slicer the tokens are stored?

WaresWichall, 2024-10-23:
@duckida that is some other platform of theirs, the OAUTH login URL is uc.makeronline.com but that is protected by a captcha and is only used to grant a token. This project stopped using OAUTH when I switched the login method from email/password to tokens.

Tokens are then used to connect to `cloud-universe.anycubic.com` and `mqtt-universe.anycubic.com`.

To use a slicer token the headers in this code would need to be fixed to match whatever the slicer sends to those URLs

On windows the slicer tokens will probably stored somewhere in appdata or possibly the registry

duckida, 2024-10-23:
On Mac it is stored at /Users/username/Library/Application Support/AnycubicSlicer/AnycubicSlicer.ini there is a token field

WaresWichall, 2024-10-23:
@duckida if you can sniff the headers the slicer sends I might be able to help with an update.

It might be easier to sniff on windows than mac. 'Fiddler classic' is good.

WaresWichall, 2024-10-23:
@duckida I refactored the authentication functions into a class today to make it easier to understand.

There is a working method for MQTT logon using android tokens which I've used in the past but the tokens are tricky to find...

Using various reverse-engineering / proxy methods (which I won't link here), see if you can find these headers from your android app which will work in the integration:
- `XX-Device-Id`
- `XX-Token`

You need both for android logon.

https://github.com/WaresWichall/hass-anycubic_cloud/blob/v0.2.0rc9/custom_components/anycubic_cloud/anycubic_cloud_api/anycubic_model_auth.py

I've been unable to understand the slicer tokens... The slicer seems to behave differently to everything else in that it doesn't store the user token on disk - instead it stores an OAUTH `access_token` which it then exchanges for short-lived user tokens, which don't seem to allow MQTT logon

WaresWichall, 2024-10-23:
If there are any android app developers reading this feel free to message me on the anycubic discord as I know a way to overcome logon issues with a fairly simple webview app but I haven't got the time to do this myself.

adamoutler, 2024-10-24:
Have you tried MITM?  I haven't but I could if you'd like a copy of the traffic.

Cloud500, 2024-10-25:
> Have you tried MITM? I haven't but I could if you'd like a copy of the traffic.

I would join this, I wanted to see (hopefully this weekend) what is "running through the wire" in LAN mode anyway, could also take a look at communication with the cloud.

WaresWichall, 2024-10-25:
I have a working implementation of the REST API using slicer `access_token`s but I haven't worked out MQTT logon yet.

See here for updated notes: https://github.com/WaresWichall/hass-anycubic_cloud/commit/0703474e1823e6a34e634fc349be72f002c8728d

I haven't added slicer logon as a homeassistant option yet since there's no added benefit.

MITM for MQTT won't help much, I know what the logon payloads from the slicer look like but there's a key ingredient in how it hashes tokens that needs further investigation of slicer code (using reverse engineering tools).

The missing piece is documented here:
https://github.com/WaresWichall/hass-anycubic_cloud/blob/0703474e1823e6a34e634fc349be72f002c8728d/custom_components/anycubic_cloud/anycubic_cloud_api/anycubic_model_auth.py#L257-L263

WaresWichall, 2024-10-29:
Good news :)

https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.2.1rc1

I've pushed a pre-release with token slicer `access_token` authentication

On windows this token is found here for slicer Next:
```
%AppData%\AnycubicSlicerNext\AnycubicSlicerNext.conf
```
or
```
C:\Users\<USERNAME>\AppData\Roaming\AnycubicSlicerNext\AnycubicSlicerNext.conf
```

Copy the whole string without the quotes:

[image]

Reconfigure home-assistant and let me know how you get on :)

Once you have hass configured I suggest you clear the `access_token` field in the slicer config (set to `""`) otherwise you'll log yourself out on each one and get errors each time they fetch updates

WaresWichall, 2024-10-29:
If anyone is interested what happened under the hood:

https://github.com/WaresWichall/hass-anycubic_cloud/blob/43068f4a4ad37d01e93fecd710bc455cb5bc0a6d/custom_components/anycubic_cloud/anycubic_cloud_api/anycubic_model_auth.py#L301-L311

WaresWichall, 2024-10-30:
Realised it wasn't easy to change authentication if already set-up so pushed a new release improving the config+options flows

https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.2.1rc2

duckida, 2024-11-02:
Doesn’t work for me with Mac

WaresWichall, 2024-11-02:
@duckida you'll need to use the windows version of slicer next, I don't think anycubic updated the mac version

duckida, 2024-11-07:
oh ok

WaresWichall, 2024-11-22:
Lots more fixes in the latest pre-release: https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.2.2b4

Has anyone else tried slicer token auth yet?

bongiozzo, 2025-02-11:
Hello!

Wrote to Anycubic support:

---

I'd like to have ability to integrate Anycubic to my Home Assistant
There is nice project 
https://github.com/WaresWichall/hass-anycubic_cloud

But maintainer wrote that you limited API, reset sessions when they are copied from Slicer Next (bad scenario anyway).

How I can manage printing with the best smart home platform (for instance turn off power after printing or send time lapse in the process to my Telegram) with Anycubic in future?
What are you going to do with open standards?

I was going to upgrade to Kobra S1 from Kobra 3, but Open API is key factor of choosing platform.

Thinking about moving to open Creality devices.

Thank you in advance.

filemondr, 2025-03-02:
Good afternoon!
I try to keep my Home Assistant allways updated (I know my bad :) ), and i got an error that the intagration can not loaded. 
The problem is that the intergration need "paho-mqtt==1.6.1", but i cant install ,because have newer version installed.
How i fixed:
1.Go to custom_components/anycubic_cloud/  (I used the file editor)
2.Open manifest.json
3."requirements": ["paho-mqtt==1.6.1"] change to "requirements": ["paho-mqtt=2.1.0"]
Restarted HA and all worked.
I hope saved some time for others who using it ;)

SHROOM246, 2025-03-20:
you sir are a legend! 

I hate to be that guy but I could only get it to work after putting "requirements": ["paho-mqtt==2.1.0"] were as you had typed "requirements": ["paho-mqtt=2.1.0"] the double == seems to make a difference if anyone else it stuck

Federbein0815, 2025-03-29:
> Hello!
>
> Wrote to Anycubic support:
>
> I'd like to have ability to integrate Anycubic to my Home Assistant There is nice project https://github.com/WaresWichall/hass-anycubic_cloud
>
> But maintainer [...]

@bongiozzo did you get already an answer from support?
Best regards

Lowcivicman99, 2025-06-13:
I've changed mine but it's been so long since this has worked mine has logged out. I've tried to reauthenticate but it's not working, anyone else having the same issue?

duckida, 2025-06-13:
Mine just works - using Slicer Next.

Gamer08YT, 2025-07-04:
> Good afternoon! I try to keep my Home Assistant allways updated (I know my bad :) ), and i got an error that the intagration can not loaded. The problem is that the intergration need [...]

Thank you for the tip ;)