WaresWichall/hass-anycubic_cloud #37: Authentication failed for registration

keidischoi, 2024-11-07:
I registered with home assistant, but it doesn't work as shown in the image. I tried using the Slicer method, but there is no token in the path setting file.
[image]

WaresWichall, 2024-11-07:
@keidischoi are you using the pre-release `v0.2.1rc3` ?

Have you logged in with the slicer? Which version of the slicer are you using?

Can you please translate the text in that image for me?
Thanks

WaresWichall, 2024-11-07:
@keidischoi sorry I think this is actually an issue with the model of printer you have, which is untested. Which printer do you have?

WaresWichall, 2024-11-07:
I think I've fixed this bug @keidischoi 

Try this release: [v0.2.1rc4](https://github.com/WaresWichall/hass-anycubic_cloud/releases/tag/v0.2.1rc4)

Let me know about my above slicer questions so we can get that working for you too.

keidischoi, 2024-11-08:
Sorry for the late reply. It works fine after upgrading to v0.2.1rc4. My model is M7 Pro. Thank you.

WaresWichall, 2024-11-08:
Great thanks for letting me know :) did you manage to find your slicer token too?

closed 2024-11-09 by WaresWichall

keidischoi, 2024-11-10:
I still can't find anything about slicer tokens, but that's fine with me since I don't use them.
tanks you.

%AppData%\AnycubicSlicerNext\AnycubicSlicerNext.conf
{
    "anycubic_presets": [
        {
            "curr_bed_type": "3",
            "filament": "Anycubic PLA @Anycubic Kobra 2 0.4 nozzle",
            "filament_colors": "#69A4FF",
            "flush_volumes_matrix": "0.000000",
            "flush_volumes_vector": "140.000000|140.000000",
            "machine": "Anycubic Kobra 2 0.4 nozzle",
            "process": "0.20mm Standard @Anycubic Kobra 2 0.4 nozzle"
        },
        {
            "curr_bed_type": "3",
            "filament": "Default Filament",
            "filament_colors": "#69A4FF",
            "flush_volumes_matrix": "0.000000",
            "flush_volumes_vector": "140.000000|140.000000",
            "machine": "Default Printer",
            "process": "Default Setting"
        }
    ],
    "app": {
        "allow_ip_resolve": true,
        "anycubic_china_tag": false,
        "anycubic_test_env": false,
        "associate_3mf": false,
        "associate_step": false,
        "associate_stl": false,
        "auto_calculate": true,
        "auto_calculate_when_filament_change": true,
        "backup_interval": "10",
        "backup_switch": true,
        "camera_navigation_style": "0",
        "check_stable_update_only": false,
        "curr_bed_type": "3",
        "dark_color_mode": "0",
        "default_page": "1",
        "developer_mode": false,
        "download_path": "C:\\Users\\keidis\\Downloads",
        "drop_project_action": true,
        "enable_multi_machine": false,
        "enable_ssl_for_ftp": true,
        "enable_ssl_for_mqtt": true,
        "flush_multiplier": "1.000000",
        "internal_developer_mode": false,
        "iot_environment": "3",
        "keyboard_supported": "none/alt/control/shift",
        "language": "en_GB",
        "last_backup_path": "C:\\Users\\keidis\\AppData\\Local\\Temp/anycubicslicer_model/Thu_Sep_26/14_43_58#49520#34",
        "log_severity_level": "warning",
        "max_recent_count": "18",
        "max_send": "3",
        "mouse_supported": "mouse left/mouse middle/mouse right",
        "mouse_wheel": "0",
        "move_view": "none/mouse left",
        "pc_device_id": "3E7CE1062716EE8F72AA0A58F14CB46E",
        "precise_control": "none/mouse left",
        "preset_folder": "",
        "privacy_version": "00.00.00.00",
        "region": "Global",
        "remember_printer_config": true,
        "reverse_mouse_wheel_zoom": false,
        "rotate_view": "none/mouse left",
        "sending_interval": "5",
        "show_3d_navigator": true,
        "show_build_edgets": false,
[... 70 lines cut ...]
            "nozzle_diameter": "0.4",
            "vendor": "Anycubic"
        },
        {
            "model": "Anycubic Kobra 2 Pro",
            "nozzle_diameter": "0.4",
            "vendor": "Anycubic"
        },
        {
            "model": "Anycubic Kobra 3",
            "nozzle_diameter": "0.4",
            "vendor": "Anycubic"
        }
    ],
    "presets": {
        "filaments": null,
        "machine": "Anycubic Kobra 2 0.4 nozzle"
    },
    "print": {
        "bed_leveling": "1",
        "flow_cali": "1",
        "timelapse": "1"
    }
}
# MD5 checksum EF4EA335DC9BF582ADE508162DD788CD

PR #36 (2024-11-09): v0.2.1 - Improvements
# v0.2.1 Changes

## General
- Improved handling of server errors during server maintenance / downtime.
- Authentication refactoring + more methods supported
- Disable MQTT connection attempts when using WEB authentication
- Add menu to `re-configure` flow:
  - Re-authenticate
  - Change Printer(s)
- Add menu to  `options` configure flow:
  - MQTT Settings
  - Drying Presets (If supported)
  - Panel Card Config
  - Debug Settings
- Fix bug introduced in `0.2.0` for printers without `parameter` property e.g. resin ( Closes #37 )