WaresWichall/hass-anycubic_cloud #41: Home Assistant not working after latest update

keidischoi, 2025-03-06:
After updating to the latest version of home assistant, hass-anycubic_cloud does not work. I tried entering a new token, but it did not work.

Core 2025.3.0
Supervisor 2025.03.0
Operating System 14.2
Front-end 20250305.0

Initially, 3 error logs were displayed, but now only 1 log is visible.
Setup failed for custom integration 'anycubic_cloud': Requirements for anycubic_cloud not found: ['paho-mqtt==1.6.1'].

rutj87, 2025-03-06:
Also experiencing same issue

keidischoi, 2025-03-06:
> 처음

It works. Thank you.

DieMetRik, 2025-03-06:
> to
> "requirements": ["paho-mqtt"]
Thanks!

LarsBorris, 2025-03-06:
> At first glance, this solution worked for me:
> 
> In the custom_components/anycubic_cloud/manifest.json file, I changed this line:
> 
> "requirements": ["paho-mqtt==1.6.1"]
> 
> to
> 
> "requirements": ["paho-mqtt"]
> 
> This allows the latest available version to be accepted.

Works perfectly! Thanks!

duckida, 2025-03-08:
#### Description
The current version of `hass-anycubic_cloud` pins `paho-mqtt` to version `1.6.1`. This practice is causing issues with the dependency graph and interferes with Home Assistant’s installed version of it.

#### Proposed Solution
To address this issue, it is recommended to specify a version range for `paho-mqtt` instead of pinning it to a specific version. This will allow for greater flexibility in dependency resolution and reduce potential conflicts with other packages.

```yaml
# Example of specifying a version range
paho-mqtt>=1.6,<2.0
```

WaresWichall, 2025-03-12:
Hi everyone, sorry for not responding sooner on this and other issues, I've had severe health problems since the end of last year.

I won't be able to test this very soon myself, can anyone confirm if this fix is working with MQTT login as well?

duckida, 2025-03-15:
Hi @WaresWichall, I can confirm MQTT mode works fine with my version! Also, the new AnycubicSlicerNext for macOS also supports the access token, so we don’t need only Windows!