WaresWichall/hass-anycubic_cloud #49: Not an issue - Kobra S1 Combo + Ace - Compatibility confirmation

FredyReady, 2025-04-01:
As title states, I can confirm Kobra S1 Combo + Ace unit compatibility, token from AnycubicSlicerNext.conf.

PsY4, 2025-04-24:
I can confirm it's working flawlessly ! 
It would be great if someone could add the box light control as the cherry on the cake ;)

pakhochen, 2025-06-09:
> As title states, I can confirm Kobra S1 Combo + Ace unit compatibility, token from AnycubicSlicerNext.conf.
> 
> Thanks author for the great work!

Can you use these buttoms?

[image]

lauer, 2025-06-09:
> > As title states, I can confirm Kobra S1 Combo + Ace unit compatibility, token from AnycubicSlicerNext.conf.
> > Thanks author for the great work!
> 
> Can you use these buttoms?

I cant :/