WaresWichall/hass-anycubic_cloud #50: Server Error Server got itself in trouble

tobiasbrage, 2025-04-05:
Every time I try to add the integration I get the error 500 Internal Server Error Server got itself in trouble.
Tried restarting and reinstall the integration with no luck.

Running HA in a docker container.

unkn0wnr0xx, 2025-04-07:
https://github.com/WaresWichall/hass-anycubic_cloud/pull/44/commits/363322807e3c86b43ca6e43bfd8d6d1c2d5a9183

tobiasbrage, 2025-04-07:
@unkn0wnr0xx thanks for commenting. It links to a code fixing the issue? How do I implement it?
Thanks.

FraatGIT, 2025-04-19:
I have the same issue, is there a solution?

duncanch89, 2025-04-21:
@tobiasbrage @FraatGIT 

Here's what I did, which worked:

1. open the link @unkn0wnr0xx posted and copy lines 15-19 from the code snippet on that page

2. navigate to your config folder on your home assistant instance

3. in your home assistant instance, open the file /homeassistant/custom_components/anycubic_cloud/manifest.json

4. paste the lines you copied into the file (replacing the existing, similar lines)

5. save the file, reboot home assistant, then try again